smart-pix/asic-testing-fnal-lab-notebooks #4: 06/14/24

badeaa3, 2024-06-14:
Goals:
- [ ] Why ZCU is outputting 10 MHz instead of 400 MHz?
- [ ] Set up memory map and spacely to write and read from Cristian's firmware

badeaa3, 2024-06-14:
We had a major problem that the mezzanine was not receiving power. This halted all day's work. It turns out that the problem was that we needed to boot the FPGA without the mezzanine attached first.

[image]

badeaa3, 2024-06-14:
The current picture status of the FPGA.
[image]

The following pictures are of the scope when we move the scope probes to different probes on the mezzanine J1 slot.

- Mezzanine J1, Pin 37 = 400 MHz 
[image]

- P9 = 100 MHz
[image]

- P1 BXCLK_ANA = 40 MHz
[image]

- P5 BXCLK = 40 MHz
[image]

- The reference is BXCLK_ANA. With respect to the rising of edge of BXCLK_ANA, the rising edge of BXCLK is shifted either positively or negatively. In this picture it's shifted positively two units, which is 2*2.5ns = 5 ns. The 2.5ns is the period of the 400 MHz clock, on which the firmware is running and the 40 MHz clocks are derived from. 
[image]

bparpillon, 2024-06-14:
[image]
We need to comment line 151 which brute force program the FW register into IP2, with x10 divider,

bparpillon, 2024-06-14:
We are unable to program the FPGA clock directly from the Zynq IP - this bug is reported in this forum and seems to be linked to Petalinux:
https://support.xilinx.com/s/question/0D52E00006iHm6zSAC/enabling-pl-clocks-in-zynq-mpsoc?language=en_US

As a workaround we inserted the clock wizard IP and are generating  the fast clock that way

badeaa3, 2024-06-15:
Other useful links I found yesterday:
- [A Data Acquisition System
for the CLIC Vertex Detector Readout Chip](https://indico.cern.ch/event/608587/contributions/2614190/attachments/1521398/2376854/A.Fiergolski_CaRIBOu.pdf). This is from 2017 using caribou and a ZCU706 inside of a physical test beam setup
- [Signal and pinout of FMC mezzanine](https://fmchub.github.io/appendix/VITA57_FMC_HPC_LPC_SIGNALS_AND_PINOUT.html)
- [Xilinx system controller to set manually VADJ](https://support.xilinx.com/s/article/67308?language=en_US)