smart-pix/asic-testing-fnal-lab-notebooks #45: Idea for Using Hardware to Inform ML Training

badeaa3, 2024-11-19:
Core of the idea -- use the physical chip during the training of the network or at a fine tuning step 

During training:
- Push the weights to the chip, do many injections, get the DNN result from chip
- Add term into loss function that encourages network to find weights that yield correct results on chip

badeaa3, 2024-11-20:
From Jennet: Accuracy at the qkeras level on the test dataset with no reweighting to a physical pT distribution = 0.7549742394504416

badeaa3, 2024-12-05:
to start on the ML training we need to do the following:
- compare the on chip results with the expected results for each input, rather than the RTL evaluated on the read back y-profile. This will give us an estimate of how much the “chip + testbench system” impacts the NN predictions. We need to quantify exactly what we aim to improve
- implement a spacely routine by copy-pasting the quantized training code from other ppl’s python scripts. run it without interfacing with the chip
- add into the function an intermediary step where the current training weights and inputs are sent to the chip. then we read back the chip predictions. add a term to the loss function that utilizes the readback. we could start with a MSE between the chip predictions (noisy) and the python output (idealized case), which the training should minimize. We will likely need to regulate/study it to make sure we dont end up at local (stupid) minima like all 0 weights

badeaa3, 2024-12-05:
Training code:
- https://github.com/smart-pix/smart-pixels-ml/blob/main/train_model.py
Model code:
- https://github.com/smart-pix/smart-pixels-ml/blob/main/models.py

bparpillon, 2024-12-06:
**follow up  idea:**
We should  inject 100nA of current on _Ileak_ input to model a high radiation dose on every pixels.
Then we retrain the DNN and assess if the network can correct

badeaa3, 2024-12-06:
Virtual environment to run trainings with: `source /asic/projects/C/CMS_PIX_28/testing/smartpixenv/bin/activate`

badeaa3, 2024-12-06:
Cernbox link with datasets: https://cernbox.cern.ch/s/IAtFmNmA9qFGHPt

badeaa3, 2024-12-06:
- Information about the different csv files: https://github.com/GiuseppeDiGuglielmo/directional-pixel-detectors/commit/c92fab36e469ecc8f235c6db243d9889a78a04f3
- Simplified running of model: https://github.com/GiuseppeDiGuglielmo/directional-pixel-detectors/blob/asic-flow/multiclassifier/SimplifiedTest.py#L78-L117

badeaa3, 2024-12-12:
Training within spacely successful!
```
# Spacely ready to take commands (see "help" for details)
<DBG> <2024-12-11 22:33:47> Unmonitored Rails: ['VTH2', 'VTH1', 'VTH0', 'VMC', 'SUPERPIX', 'INJ_1']
> ~r7
<DBG> <2024-12-11 22:33:52> Evaluating: ROUTINE_DNNTraining()
Model: "model"
_________________________________________________________________
 Layer (type)                Output Shape              Param #
=================================================================
 input1 (InputLayer)         [(None, 16)]              0

 dense1 (QDenseBatchnorm)    (None, 58)                1219

 relu1 (QActivation)         (None, 58)                0

 dense2 (QDense)             (None, 3)                 177

 linear (Activation)         (None, 3)                 0

=================================================================
Total params: 1396 (5.46 KB)
Trainable params: 1279 (5.00 KB)
Non-trainable params: 117 (472.00 Byte)
_________________________________________________________________
Interpreting Model
Topology:
Layer name: input1, layer type: InputLayer, input shapes: [[None, 16]], output shape: [None, 16]
Layer name: dense1, layer type: QDenseBatchnorm, input shapes: [[None, 16]], output shape: [None, 58]
Layer name: relu1, layer type: Activation, input shapes: [[None, 58]], output shape: [None, 58]
Layer name: dense2, layer type: QDense, input shapes: [[None, 58]], output shape: [None, 3]
Layer name: linear, layer type: Activation, input shapes: [[None, 3]], output shape: [None, 3]
Interpreting Model
Topology:
Layer name: input1, layer type: InputLayer, input shapes: [[None, 16]], output shape: [None, 16]
Layer name: dense1, layer type: QDenseBatchnorm, input shapes: [[None, 16]], output shape: [None, 58]
Layer name: relu1, layer type: Activation, input shapes: [[None, 58]], output shape: [None, 58]
Layer name: dense2, layer type: QDense, input shapes: [[None, 58]], output shape: [None, 3]
Layer name: linear, layer type: Activation, input shapes: [[None, 3]], output shape: [None, 3]
Creating HLS model
Writing HLS project
/asic/projects/C/CMS_PIX_28/testing/tools/venvs/p3.11.11/lib/python3.11/site-packages/keras/src/engine/training.py:3000: UserWarning: You are saving your model as an HDF5 file via `model.save()`. This file format is considered legacy. We recommend using instead the native Keras format, e.g. `model.save('my_model.keras')`.
  saving_api.save_model(
Done
     0      1      2
0  0.0 -0.375  0.375
(13982, 16) (13982,)
Epoch 1/10
11/11 [==============================] - 2s 39ms/step - loss: 0.5970 - sparse_categorical_accuracy: 0.7326 - val_loss: 0.6050 - val_sparse_categorical_accuracy: 0.7404
Epoch 2/10
11/11 [==============================] - 0s 5ms/step - loss: 0.5997 - sparse_categorical_accuracy: 0.7344 - val_loss: 0.6073 - val_sparse_categorical_accuracy: 0.7358
Epoch 3/10
11/11 [==============================] - 0s 4ms/step - loss: 0.5998 - sparse_categorical_accuracy: 0.7324 - val_loss: 0.6352 - val_sparse_categorical_accuracy: 0.7143
Epoch 4/10
11/11 [==============================] - 0s 4ms/step - loss: 0.6115 - sparse_categorical_accuracy: 0.7273 - val_loss: 0.6518 - val_sparse_categorical_accuracy: 0.6972
Epoch 5/10
11/11 [==============================] - 0s 4ms/step - loss: 0.6019 - sparse_categorical_accuracy: 0.7305 - val_loss: 0.6247 - val_sparse_categorical_accuracy: 0.7354
Epoch 6/10
11/11 [==============================] - 0s 4ms/step - loss: 0.5989 - sparse_categorical_accuracy: 0.7321 - val_loss: 0.6240 - val_sparse_categorical_accuracy: 0.7347
Epoch 7/10
11/11 [==============================] - 0s 4ms/step - loss: 0.6103 - sparse_categorical_accuracy: 0.7287 - val_loss: 0.6404 - val_sparse_categorical_accuracy: 0.7261
Epoch 8/10
11/11 [==============================] - 0s 4ms/step - loss: 0.5988 - sparse_categorical_accuracy: 0.7377 - val_loss: 0.6178 - val_sparse_categorical_accuracy: 0.7397
Epoch 9/10
11/11 [==============================] - 0s 4ms/step - loss: 0.5896 - sparse_categorical_accuracy: 0.7464 - val_loss: 0.6153 - val_sparse_categorical_accuracy: 0.7272
Epoch 10/10
11/11 [==============================] - 0s 4ms/step - loss: 0.6048 - sparse_categorical_accuracy: 0.7367 - val_loss: 0.6164 - val_sparse_categorical_accuracy: 0.7372
/asic/projects/C/CMS_PIX_28/testing/tools/venvs/p3.11.11/lib/python3.11/site-packages/keras/src/engine/training.py:3000: UserWarning: You are saving your model as an HDF5 file via `model.save()`. This file format is considered legacy. We recommend using instead the native Keras format, e.g. `model.save('my_model.keras')`.
  saving_api.save_model(
Save: model.h5
437/437 [==============================] - 1s 1ms/step - loss: 0.6016 - sparse_categorical_accuracy: 0.7429
Test loss: 0.6015514135360718
Test accuracy: 0.7428836822509766
<INF> <2024-12-11 22:34:00> This Routine took: 0:00:08
```