smart-pix/asic-testing-fnal-lab-notebooks #52: 12/10/24 Analog Code Cleanup #1

badeaa3, 2024-12-10:
- Finish cleaning up code to S-curve
- Implement get.stream  (this will require some changes in the way we readback the data)
- take S-curve data
- make figure ready for paper

bparpillon, 2024-12-12:
### Implementation of stream_memory 

**First we need to update the Peary server to incorporate the new function in aq_dev branch**
**scp peary folder to FPGA**

```
export PATH=/home/petalinux/temp/cmake-3.29.0-rc4-linux-aarch64/bin/:$PATH
cd peary
mkdir build
cd build
cmake -DBUILD_SpacelyCaribouBasic=ON -DBUILD_CMSPIX28=ON -DINTERFACE_IIO=OFF -DINTERFACE_MEDIA=OFF -DINTERFACE_I2C=ON ..
make -j4
make install
```

badeaa3, 2024-12-12:
Upgrading peary
```
# On host machine
git clone https://gitlab.cern.ch/adquinn/peary.git # clone adam's peary
cd peary
git checkout aq_dev # go to dev branch
cd ../

# scp peary folder to FPGA
scp -r peary/ petalinux@192.168.1.24:/home/petalinux/tools/

# log onto FPGA
ssh petalinux@192.168.1.24

# setup cmake (assumed you already scp'd it to the fpga)
export PATH=/home/petalinux/temp/cmake-3.29.0-rc4-linux-aarch64/bin/:$PATH

# compile peary
cd /home/petalinux/tools/peary
mkdir build
cd build
cmake -DBUILD_SpacelyCaribouBasic=ON -DBUILD_CMSPIX28=ON -DINTERFACE_IIO=OFF -DINTERFACE_MEDIA=OFF -DINTERFACE_I2C=ON ..
make -j4
make install
```

badeaa3, 2024-12-12:
Performance plot from Adam 
[image]

CodexForster, 2025-10-20:
Since the new ASIC utilizes more bias lines from the carboard, we need to edit aq_dev's peary code. Add these lines to `peary/devices/SpacelyCaribouBasic/SpacelyCaribouBasicDevice.cpp` after line 51 (`periphery.add("BIAS_5", carboard::BIAS_5);`):
```
periphery.add("BIAS_26", carboard::BIAS_26);
periphery.add("BIAS_15", carboard::BIAS_15);
periphery.add("BIAS_6", carboard::BIAS_6);
periphery.add("BIAS_7", carboard::BIAS_7);
periphery.add("BIAS_9", carboard::BIAS_9);
periphery.add("BIAS_13", carboard::BIAS_13);
```