smart-pix/asic-testing-fnal-lab-notebooks #55: 12/17/24 - TO DO AFTER THE BREAK

bparpillon, 2024-12-17:
### DNN OPTIMIZATION

- [ ] dnn_csv load takes 0.2 to 0.4 sec and should be optimized : maybe save the csv in the SDcard
- [ ] shift register programming requires more than 10,000 writes operations, this could be optimized by creating a stream_set_memory command
- [ ] scanchain readback on DATA_ARRAY_0 can be optimized by NOT extracting the y-profile thus saving 24 read instructions. The argument **readYproj** should be set to **False**
- [ ] scanchain readback on DATA_ARRAY_1 can also be optimized by NOT reading the BxCLK data. A FW upgrade would be necessary to optimize the dnn_output data and compress the essential bits in a single word.

### TRAINING OPTIMIZATION
- [ ] Simplify the DNN evaluation flow
- [ ] Feature space to explore
1. alpha
2. send each pattern 10 times to **guide** the loss optimization against noise 
3. Enlarge the training and separate from eval set

bparpillon, 2024-12-18:
**To extract the new weight/csv file:**
1. update this line with the model file https://github.com/smart-pix/CMSPIX28Spacely/blob/main/CMSPIX28Spacely_Subroutines_B5_DNNTraining.py#L417
2. uncomment it and comment out the previous line
3. set this to false https://github.com/smart-pix/CMSPIX28Spacely/blob/main/CMSPIX28Spacely_Subroutines_B5_DNNTraining.py#L438
4. then run the file `python CMSPIX28Spacely_Subroutines_B5_DNNTraining.py`

badeaa3, 2024-12-26:
Readings to brainstorm ideas:
- [HAQ: Hardware-Aware Automated Quantization with Mixed Precision](https://arxiv.org/abs/1811.08886)