smart-pix/asic-testing-fnal-lab-notebooks #62: 20250312 - Goals

bparpillon, 2025-03-12:
**** ### Goals

- [ ] Check vin_test amplitude in the board

badeaa3, 2025-03-12:
Add protection to imports so that routines crash if imports don't work properly and give the user a large message
```
import sys

def loud_error(message):
    print("\033[91;1m" + "="*80)  # Red, bold, and large separator
    print(" " * 20 + "IMPORT ERROR!")  
    print("="*80)
    print("\033[93;1m" + message.upper() + "\033[0m")  # Yellow, bold text
    print("\033[91;1m" + "="*80 + "\033[0m")  # Reset color

try:
    import numpy as np  # Example import
    import tensorflow as tf
except ImportError as e:
    loud_error(str(e))
    sys.exit(1)  # Exit script immediately
```

bparpillon, 2025-03-13:
**Other goals:** 

- [ ] Talk to Tom and do PCB review
- [ ] Get an update of the breakout boards
- [ ] Ask buddy Bryan for help