smart-pix/asic-testing-fnal-lab-notebooks #72: 20250414 - Goals

bparpillon, 2025-04-14:
### Hands On Test

**Meetings/Discussion**

- [ ] Discuss with Cristian how to address some of the changes, as there are multiple route we could take. For instance scanLoad could be aligned with falling edge of clock and left as is (no changes with test_delay).
- [ ] Ask Cristian also about the FW status clear code which seems to be acting out?
- [ ] Meet with Jeff for 30mn
- [ ] 1PM meeting with Jim
- [ ] Ask Tom if 0.1ohm resistance was received, if it was make the change in the carBoard

**Prepare boards and setup to ship to UC and Cornell**
- [ ] Test each breakout boards
- [ ] We will also ship the DNN cable - they need to be tuned
- [ ] Prepare filters and connection mapping

**Hands On Test**
- [ ] Observe the DNN output and delay and see if we need to make a FW change for that. Likely we should remap DNN_OUTPUT to the mezzanine
- [ ] If Cristian provide the new FW corresponding to the updated timing diagram (he said it will be quick)  - flash and check pulses on scope and adjust routines (if needed) to plot a single Scurve
- [ ] Check 2D plot of a single pixel scanchain matches with the serial 2D plot (it really should)

bparpillon, 2025-04-14:
### List of things to send to UC and Cornell:

- [ ] 1 - Tested Breakout boards
- [ ] 1 - Long FMC Cable
- [ ] 1 - Analog filters on Breadboards (VMC and VTH)
- [ ] 4 - Power cable (mini banana plugs)
- [ ] 2 - DNN_OUTPUT cables with discriminator pre-vetted 
- [ ] Wires or solution to map
- [ ] ASIC board (Cornell Only)

bparpillon, 2025-04-14:
DNN Cable tune up:

1. Connect DISC_LVDS to ASIC board
2. Connect LVDS_CMOS to scope with input set to 50 ohm load
3. Set power on LVDS_CMOS side to 5V
4. Set scope Trigger to Auto mode and free running
5. Tune the potentiometer with a screwdriver until the signal goes high = 0.9V
6. Set scope Trigger to Mode normal, connect vin_test to another input (AC coupled) and trigger on it with falling edge
7. Run DNN routine (~r6) and dnn_output<0> should go from 1 to 0

badeaa3, 2025-04-14:
Preparing boxes to send to schools. 
- Table of DUTs and their current status https://github.com/smart-pix/asic-testing-fnal-lab-notebooks/issues/57#issuecomment-2679942183

bparpillon, 2025-04-17:
### ### **SETTINGS PREDICTION**
**test_delay** = 0x03
**BxCLK_DELAY**  = 0x0B
BxCLK period = 100ns or 0x28 setting
Half **BxCLK** period is 0x14

found scanLoadPhase = 0x20 to be aligned with falling edge of the clock
this corresponds to 0x0B + 0x14 = 0x1F
This matches what we find in experiment (0x20)

bparpillon, 2025-04-17:
| scanLoadPhase | ScanLoad rising edge to BxCLK rising edge delay | 
|:-----------|:------------:|
| 1          |     -27.5ns       |   
| 8          |     -107.5ns       |        
| 10          |     -87.5       |   
| 18          |     -70.0ns       |        
| 20          |     -47.5ns       |   
| 28          |     -28.5ns       |

bparpillon, 2025-04-17:
We measured the debug point to observe where the bit0 will be sampled (there are 0x28 possible test_sample settings)

We observed that test_delay push that window to the right and is very important to increase the sampling window.

In green : scanOut signal (mezzanine)
In magenta: debug point bit0 capture window (mezzanine)
in yellow: injection (ASIC)

test_delay = 0x03
test_sample = 0x08
BxCLK_delay = 0x0B

[image]

bparpillon, 2025-04-18:
### 2D plot 

**bit255 (bit0 is the same)
BxCLK_DELAY = '0B
scanLoadPhase = '20'
nsample = 100**

[image]

bparpillon, 2025-04-18:
### 2D plot- bit255 - testing injection delay

(bit0 is the same)
BxCLK_DELAY = '0B
scanLoadPhase = '20'
nsample = 100
injDly = 'F'

[image]

bparpillon, 2025-04-24:
**Firmware Change  -  20250423** This FW has the new modification for IP2 test 3, 4 and 5.

- [ ] Flash the FPGA and check that everything works as before for IP2 test 2
- [ ] Test IP2 test 4
- [ ] Test IP2 test 5