smart-pix/asic-testing-fnal-lab-notebooks #76: 05/07/25

badeaa3, 2025-05-07:
Goals:
- Merge request for subroutines
- Work on filtering the secondary gaussians for analysis of new SCurve data with vth0 set at 400e-
- Validate the DNN 98% accuracy compared to RTL

badeaa3, 2025-05-07:
Possible desktop machines:
- https://www.perplexity.ai/search/what-is-the-best-desktop-machi-KQlaT39PSjul.8HNI_U6Kw
- https://www.dell.com/en-us/shop/desktop-computers/alienware-area-51-gaming-desktop/spd/alienware-area-51-aat2250-gaming-desktop/useaat2250wcto01

bparpillon, 2025-05-07:
**Path:**
/mnt/local/CMSPIX28/Scurve/data/ChipVersion1_ChipID9_SuperPix2/2025.05.06_17.07.50_MatrixNPix_vMin0.001_vMax0.400_vStep0.00100_nSample1365.000_vdda0.900_VTH1.000_BXCLKf10.00_BxCLKDly45.00_injDly75.00/

**Settings:**

badeaa3, 2025-05-07:
Title: In-pixel integration of signal processing and AI/ML based data filtering for particle tracking detectors
Abstract:
We present a smart pixel prototype readout integrated circuit (ROIC), designed in a 28 nm CMOS bulk process, featuring in-pixel implementation of an artificial intelligence (AI) / machine learning (ML) based data filtering algorithm. This serves as a proof-of-principle for future particle tracking detectors. The ROIC integrates in-pixel signal processing with a fully connected two-layer neural network (NN) that analyzes information from a cluster of 256 pixels. The NN classifies, in real time, whether an incident particle has high or low momentum based on the spatial pattern detected in the pixel array. In this talk, we will provide an overview of the ROIC architecture and present testing results from the first fabricated chip.