smart-pix/asic-testing-fnal-lab-notebooks #78: 202050512 - DNN / current measurement / SuperPix1

bparpillon, 2025-05-12:
**Goals:**

- [ ]  Plan WB with Pam:
   - Provide CLEAR layout pictures
   - Ask to WB 14 boards and save 1 to test **noise**

- [ ]  DNN debug:
   - Show Sergei the issues and discuss improvement
   - Plan better algorithm - **voting system will not work**

- [ ] Power measurement:
   -  Fix carboard current meter for at least 2 power
   - Measure and report current with analog, digital and DNN

- [ ] Meetings
   - 12PM with students
   - 1PM for chiplets 1 and 9

- [ ] Simulation results
   - Extract Qth with new design and injection test site (liberate, 30 noise run, 256 MC)
   - Extract Qth with new design and normal sensor input (liberate, 30 noise run, 256 MC)
   - Extract Qth with new design (moderate, 40 noise run, 256 MC)

badeaa3, 2025-05-12:
FFT of the scope (no signal connected to probe)

[image]

badeaa3, 2025-05-12:
FFT of vth2

[image]

badeaa3, 2025-05-12:
FFT of carboard master VTH

[image]

badeaa3, 2025-05-12:
FFT of dvss
[image]

badeaa3, 2025-05-12:
FFT of dvdd

[image]

badeaa3, 2025-05-12:
FFT of vth1

[image]

badeaa3, 2025-05-12:
Noise measurement with spectral meter.

Channel power density -138dbm/hz to -143dbm/hz

[image]

badeaa3, 2025-05-12:
We replaced on the carboard R82 from 0.01 ohm to 0.1 ohm. This line goes to power out 4, which is connected to vddd. 

vddd current is measurement by carboard to be 0.01703 A. Divide that by 10 because we increased the resistance by 10x to 0.0017 A, which is 1.7 mA. This agrees with simulation.

badeaa3, 2025-05-12:
vddd 
- 3.5V VTH, measured current 0.001764 A
- 0.5 VTH, measured current 0.033511 A

badeaa3, 2025-05-15:
Test launching new summary