smart-pix/asic-testing-fnal-lab-notebooks #88: 08/04/25 - Goals paper/cold testing/2nd test stand

bparpillon, 2025-08-04:
**Goals**

**Hands on measurements:**
- [ ] Extract a plot of leakage current vs vth values for vth0, vth1, vth2 and put it on the paper.
- [ ] Extract the power of the DNN for 10cm-2 hit density and maybe report power vs hit density
- [ ] Setup second test stand with the CarBoard pulse generator and prepare for cold testing
- [ ] Second test stand DNN testing - check discriminator
- [ ] If ASIC boards are WB - start the inventory

**Spacely:**
- [ ] Fix venv for spacely to comply with Adam's flow
- [ ] Pull latest spacely from git
- [ ] Do an MR on lab_analysis
- [ ] Do an MR on Routines

**Paper:**
- [ ] Review and clean plots to have the smart pixel format
- [ ] add one plot of cold testing to tease the reader :D

badeaa3, 2025-08-04:
To upgrade the spacey py-libs you just need to run [this line](https://github.com/SpacelyProject/spacely/blob/main/PySpacely/SetupLinux.sh#L31C1-L31C101) from with spacely/PySpacely
```
cd /asic/projects/C/CMS_PIX_28/abadea/working/CMSPIX28_DAQ/spacely/PySpacely
python -m pip install --upgrade --force-reinstall -r ./requirements/requirements-py-libs-common.txt
```

CodexForster, 2025-08-05:
[leakageCurrent.csv](https://github.com/user-attachments/files/21586236/leakageCurrent.csv)