smart-pix/asic-testing-fnal-lab-notebooks #89: 08/05/25 - DNN result paths

bparpillon, 2025-08-05:
Threshold with low noise
/mnt/local/CMSPIX28/Scurve/data/ChipVersion1_ChipID17_SuperPix1/2025.08.05_11.10.43_DNN_vth0-0.08_vth1-0.16_vth2-0.32

Threshold from filter paper: ie: 400e-, 1600e-, 2400e-
/mnt/local/CMSPIX28/Scurve/data/ChipVersion1_ChipID17_SuperPix1/2025.08.05_18.03.27_DNN_vth0-0.020_vth1-0.085_vth2-0.135

badeaa3, 2025-08-11:
**Compouts**
dataset 8
- /asic/projects/C/CMS_PIX_28/benjamin/verilog/workarea/cms28_smartpix_verification/PnR_cms28_smartpix_verification_D/tb/dnn/csv/l6/compouts.csv
- /mnt/local/CMSPIX28/data/ChipVersion1_ChipID17_SuperPix1/2025.08.07_10.02.07_DNN_vth0-0.080_vth1-0.160_vth2-0.320/

dataset 14 ylocal 0-1.35
- /asic/projects/C/CMS_PIX_28/abadea/working/filter/model_pipeline/tmp/1000_1600_2400/compouts_ylocal_0.00_1.35.csv
- /asic/projects/C/CMS_PIX_28/abadea/working/filter/model_pipeline/tmp/400_1600_2400/compouts_ylocal_0.00_1.35.csv
	
**At 150e- + [1000, 1600, 2400] e- thresholds**:
- first ~100k test vectors: /mnt/local/CMSPIX28/data/ChipVersion1_ChipID17_SuperPix1/2025.08.07_13.37.31_DNN_vth0-0.087_vth1-0.139_vth2-0.208/final_results.npy
- full ~300k test vectors: /mnt/local/CMSPIX28/data/ChipVersion1_ChipID17_SuperPix1/2025.08.08_11.00.19_DNN_vth0-0.087_vth1-0.139_vth2-0.208/final_results.npy

**Found that our calculation for electrons needed to be updated to use the CvG extraction New data taking**:
- /mnt/local/CMSPIX28/data/ChipVersion1_ChipID17_SuperPix1/2025.08.10_10.36.52_DNN_vth0-0.020_vth1-0.085_vth2-0.135

CodexForster, 2025-09-10:
Testbench 1, run 3 (99.86% match with RTL results) data-taking locations:
| Threshold [e-] | Dataset size | Location |
| ---------------- | ------------- | ------------- |
| [1000,1600,2400] | 150k | /mnt/local/CMSPIX28/data/ChipVersion1_ChipID17_SuperPix1/2025.09.09_18.06.40_DNN_vth0-0.048_vth1-0.083_vth2-0.128/ |