smart-pix/asic-testing-fnal-lab-notebooks #93: How to run cocotb

bparpillon, 2025-09-19:
**Simulation Environment –** 
We plan to use [cocotb](https://docs.google.com/presentation/d/1s9TKmZaHGkJeKYMGMMTFyUz1dQSQIiXJeF0jhEcczFE/edit?usp=sharing) for simulation.

**Machine Setup**
This setup is specifically for Fermilab ASIC servers (e.g. `fasic-beast1.fnal.gov`). The setup on personal machines may be different.

**Console Environment**
This is a one-time-only setup. Copy and paste exactly the following instructions at the end of your `~/.bashrc` file:

**Setup some local tools**
export PATH=/fasic_home/gdg/local/bin:$PATH
export LD_LIBRARY_PATH=/fasic_home/gdg/local/lib:$LD_LIBRARY_PATH

**Enable Python 3.8**
If running from fasic machine
source /opt/rh/rh-python38/enable

Close the console or ssh connection and open a new one. Try the following command:
verilator --version
Expected output: Verilator 5.008 2023-03-04 rev v5.008

Then try:
which python
# Expected output: /opt/rh/rh-python38/root/usr/bin/python

Python Virtual Environment 
This is a one-time-only setup. Run the following instructions in a new console (see previous step):
python -m venv ~/venvs/cms28-smartpix-venv
source ~/venvs/cms28-smartpix-venv/bin/activate
pip install cocotb pytest ipython fxpmath
pip install numpy
pip install pandas

Then try:
ipython

**After Machine Setup**
After the previous steps, everytime you need the virtual environment, just do:
source ~/venvs/cms28-smartpix-venv/bin/activate

**Clone Repository**
With [public/private keys](https://docs.github.com/en/authentication/connecting-to-github-with-ssh/adding-a-new-ssh-key-to-your-github-account):
git clone git@github.com:GiuseppeDiGuglielmo/cms28_smartpix_verification.git

**More setup**
Before starting any simulation make sure to activate the virtual and cadence environment:
source ~/venvs/cms28-smartpix-venv/bin/activate
source /fasic_home/gdg/bin/cadence.sh
Make sure the makefile points to the tsmc28 standard cell library.

**Run cocotb**
Go to:
cd $PATH/cms28_smartpix_verification/cms28_smartpix_test/tb/dnn
edit L37 of file cms28_pix_dnn_wrapper_ASIC_tb.py
make sim

badeaa3, 2025-09-22:
Great instructions and it worked for me!

I made a group python environment we can use. We can source this file for the environment to be setup and the simulation run: `source /asic/projects/C/CMS_PIX_28/testing/tools/setup_cocotb_rtlsim.sh`.

To update the `yprofiles.csv` update the `input_1_data` inside of `/asic/projects/C/CMS_PIX_28/testing/tools/cms28_smartpix_verification/cms28_smartpix_test/tb/dnn/cms28_pix_dnn_wrapper_ASIC_tb.py`

badeaa3, 2025-09-22:
Also linking the cocotb4test pdf from Giuseppe/Ben that is inside the fermilab Sharepoint since getting access to it is sometimes difficult.

[cocotb4test.pdf](https://github.com/user-attachments/files/22464471/cocotb4test.pdf)